ckormanyos/real-time-cpp #13: Textbook doesn't reference chapter02_03a

aswimman, 2015-01-03:
Chris - Like this textbook!  Incorporating into my course on real-time embedded software.  In Chapter 2.3, I don't see a reference to the companion code directory chapter02_03a.   Did this companion code come later after the publishing of the book?   I'm trying to determine the difference between chapter02_03 and chapter02_03a directories.   chapter02_03 does not contain the led.cpp file that is shown in the book (maybe on purpose as you suggest "Try to build, flash and run the LED program with timing...").   chapter02_03a does contain the led.cpp file that is shown as an "app" namespace.

I am running on Mac OSx and can't use Visual Studio.   I can't load the projects yet (may install Parallels so I can run VS).

Thanks!

ckormanyos, 2015-01-03:
Thank you for the kind words on the book.

Actually yes, the companion code has been refactored in some areas since publishing the book. The visible application result of Chapter02_03 and Chapter02_03a is the same --- LED on portb.5 toggles with 1/2Hz. But the timing mechanism is different.
* Chapter02_03 toggles the LED on portb.5 in the main() subroutine with a blocking one second delay between successive toggles.
* Chapter02_03a was, in fact, added *after* publishing the book. My apologies for this. This example uses a tiny multitasking scheduler to launch an LED application task which subsequently toggles the LED on portb.5 with timing via a static 1-second timer in the task. In this way, Chapterr02_03a uses material from much later in the book.

Visual Studio is not required for the target builds. Each example does, however, require the existence of a tools such as make, GCC, etc. which are not available in the Git repo. The build scripts are set up using, for example :+1: 
* chapter02_03\tools\Util\MinGW\msys\1.0\local\...
* chapter02_03a\tools\Util\MinGW\msys\1.0\bin...

Some readers simply use Mac OSx, *nix, or MinGW and supply their own paths to make, GCC, etc. in the build scripts. The builds in the examples as they appear in Git use a moved MinGW infrastructure (but not present in Git). At any rate, getting the examples up and running might require some batch, script or directory work. Get in touch if you need more details on this.

Also feel free to get back to me with any suggestions regarding the companion code, as this has been significantly improved through inputs from readers.

- Chris

aswimman, 2015-01-03:
Thank Chris for this information.   This definitely helps!   I have compiled, linked and loaded the chapter_02_02 led.cpp program and loaded using avrdude onto my Arduino Uno.   I’m using GCC 4.9.2 (from Brew) and cross-compiled to the var-unknown-elf target.

I will keep you informed if I find anything in the textbook.   My class is starting on 1/20 and I using your book in conjunction with “Real-Time Systems:   Design Principles for Distributed Embedded Applications - 2nd edition” by Kopetz.   They really go together well….

Thanks again,
Ken

ckormanyos, 2015-01-03:
Great, Ken! You are up and running.

Some errors do exist in the book and I suspect you may find others. I am working on getting an errata sheet posted. Thanks again for your interest and don't hesitate to comment or suggest improvements as you continue working with the book and the example code.

I will keep this issue open for a few days and if nothing else pops up, then close it.

Thanks again,
Chris

ckormanyos, 2015-01-10:
This this is being closed as per plan. Thanks for the initial report, Chris.

closed 2015-01-10 by ckormanyos

aswimman, 2015-01-10:
Yes - you did answer my question!

I can now report that I have gotten the four chapter problems compiled, linked on my Mac-mini (Yosemite) and on my new MacBook Pro (Yosemite) with the gcc-4.9.2_1 and then successfully loaded on my Arduino Uno via avrdude.   Next to tackle will be to do the same thing with Windows 8.1 on these computers (using Parallels) and then to get the reference app compiled, linked, and loaded.

Thanks for your help!
Ken

ckormanyos, 2015-01-10:
Nice work !