ckormanyos/real-time-cpp #15: The new Chapter09_07 source code

aswimman, 2015-03-14:
Hi Chris - I see that you've added this new section Chapter09_07 source code.   My only question is that there isn't a Chapter 9.7 in your current edition of the book?  Can you explain what this new section is for (appears to be display related)?

Thanks!

(PS. My students are enjoying the class and using your book and software....)

ckormanyos, 2015-03-17:
A second edition of the book is in progress. Thesecond edition is expected to have Sect. 9.7 contaning a full description of the seven-segment display example. Sorry, I've been a bit terse on this new stuff.

aswimman, 2015-03-18:
Chris - Great to know!  Sounds like a great example…  let me know when your drop dead date is for corrections.  We are collecting them as we go along…  Do you have an anticipated publishing date yet?   I’m going to be teaching this class again in the fall (starting in mid-August)…

Thanks!
Ken

ckormanyos, 2015-03-18:
Ken,

Please feel free to send suggestions and corrections any time---actually, the sooner the better. Thanks for your continued interest in the book and sample code.

Thanks again, Chris

ckormanyos, 2015-03-23:
Closing this issue because the original question has been addressed. New suggestions and corrections can be addressed in new issues.

closed 2015-03-23 by ckormanyos