ckormanyos/real-time-cpp #16: Question about the Reference Application

aswimman, 2015-03-14:
Chris - One of my students would like to know why the mcal_spi.h and mcal_spi.cpp are commented out in the reference application.   I told them it was because you were using the reference app to benchmark the timing of the MD5 algorithm.   Was I correct?

Thanks!

ckormanyos, 2015-03-17:
The reference application is work in progress. The SPI driver is in a very preliminary state. To this date, I do not believe that I have found a satisfactory abstraction for a low-level SPI driver. The handling of the buffers and especially the chip-select signal is, well ugly to say the least. Furthermore, I do not support SPI low-level driver for the other targets. o I just commented the preliminary AVR stuff out. The primary purpise of the reference application is to provide a skinny multitasking scheduler and many of the utilties described in the book. One of the main festures of the reference application is the ability to perform a benchmark task. Recently, I have worked more with floating-point investigations and CRC32.

aswimman, 2015-03-18:
Chris - We suspected that… that for letting us know!   You’ve answered our question (from one of my students).

Thanks!
Ken

thzoechbauer, 2015-03-23:
Hello Chris!

I spent a lot of time to compile the GNU Cross Compiler with MinGW/msys as you explained in your book. Thought it would be nice to compile my code with the newest released gcc sources :-) In your book you describe to build avr-unknown-elf. I tried to build arm-none-eabi and used the configuration of the book wherever possible (except directory names for arm-none-eabi). But at the end I wasn't able to compile a running hex with the self-compiled-cross-compiler. (My program runs if I compile with "GNU Tools for ARM Embedded Processors" from https://launchpad.net/gcc-arm-embedded.)

In the book you use GMP, MPFR, MPC, PPL, BinUtils, newlib as prerequisites. But in [target]_flags.gmk of the ref_app the prerequisites GMP, MPFS, MPC, CLOOG and ISL are mentioned. There are even some configuration-flag mismatches in the book and the ref_app (e.g.: book: --disable-shared --enable-static ... the other way round in ref_app: --disable-static --enable-shared). I'm not in building compilers and I'm a bit confused... thought this would be easier. Which configuarations are the right ones or does it make a (big) difference?

I would appreciate it if I could use the same configuration as you do and then port the code to the stm32f303 mcu mounted on the stm32f3discovery-board.

Thanks, Thomas

ckormanyos, 2015-03-23:
Building GCC from source can be a complex (at times tedious) technical endeavor. As of GCC 4.8.x, the prerequisites for the build include GMP, MPFR, MPC, CLooG and ISL (no longer PPL).

I have recently learned that the arm-none-eabi build is best done with multilib and FPU support.

Due to the complexity of this issue, I have created a dedicated issue for improving instructions for the GCC build.

ckormanyos, 2015-03-23:
I am closing this issue because I believe the original point has been addressed. Following the GCC build can be traced in the new issue.

closed 2015-03-23 by ckormanyos