ckormanyos/real-time-cpp #19: Improve instructions for the GCC build

ckormanyos, 2015-03-23:
Improve the instructions for the GCC build from source in MinGW / msys. Attempt to keep the prerequisites up-to-date. Also provide command line options for building arm-none-eabi with multilib and FPU support.

ajneu, 2016-02-07:
Have you tried MSYS2?

It's fantastic. It uses a linux package manager called pacman. Quite the Linux experience under windows!
https://msys2.github.io/

After installation, one can lauch the msys shell like this:
`C:\msys32\mingw32_shell.bat`           (for 32-bit Windows)
`C:\msys32\mingw64_shell.bat`             (for 64-bit Windows)

Search and install packages:
```bash
## search for gcc
pacman -Ss gcc

## install gcc for 32bit windows
pacman -S mingw32/mingw-w64-i686-gcc

## install gcc for 64bit windows
pacman -S mingw64/mingw-w64-x86_64-gcc

## search for make
pacman -Ss make

## install make (for 32bit windows)
pacman -S mingw32/mingw-w64-i686-make

## install make (for 64bit windows)
pacman -S mingw64/mingw-w64-x86_64-make

## search for boost
pacman -Ss boost

## install boost (for 32bit windows)
pacman -S mingw32/mingw-w64-i686-boost

## install boost (for 32bit windows)
pacman -S mingw64/mingw-w64-x86_64-boost

## search for git
pacman -Ss git

## install git
pacman -S msys/git

## search for gdb
pacman -Ss gdb

## install gdb (for 32bit windows)
pacman -S mingw32/mingw-w64-i686-gdb   mingw32/mingw-w64-i686-gdbm

##  install gdb (for 64bit windows)
...
```

ckormanyos, 2016-02-08:
Thanks for the suggestion.

ajneu, 2016-02-09:
The really nice thing about MSYS2 is that it has very recent version of compilers (better support for C++14) and tools:
Currently
* gcc 5.3.0
* boost 1.60

In addition to commands mentioned in top post, a good base system can be installed with 
```bash
pacman -S base-devel
```

(Unlike the old msys/mingw, MSYS2 is also not "dead": http://stackoverflow.com/a/25023611/5106243 )

ajneu, 2016-02-11:
If you're behind a proxy, then for networking (necessary for `pacman` and command-line `git`):

When launching 
`C:\msys32\mingw32_shell.bat (for 32-bit Windows)`
`C:\msys32\mingw64_shell.bat (for 64-bit Windows)`

be sure to set:
```bash
export http_proxy=http://proxyhost:port     ## e.g. export http_proxy=http://192.1.1.7:3128
export https_proxy=http://proxyhost:port    ## e.g. export https_proxy=http://192.1.1.7:3128
```
or; if username/password is necessary as well then:
```bash
export http_proxy=http://username:password@proxyhost:port
export https_proxy=http://username:password@proxyhost:port
```

(Also... you may wish to look at cntlm)

ajneu, 2016-02-17:
Actually... the AVR gcc crosscompiler should be packaged! That would be great!
I don't currently have time with it, but if anybody has, then I'd recommend talking to the MSYS2 people, to get the AVR cross-compiler packaged into MSYS2's pacman format. I'm sure those MSYS2 people would help.

Main benefit: no more time wasted on compiling. Less people compiling == Less energy usage. ;)

jhalbrecht, 2016-02-17:
Generally GitHub issues aren't the place for 'Me too' however with the low volume of traffic here I'm waving my hand high saying 'Me too, me too!' Maybe tweet this out and get some momentum built for the idea. I like the idea of having it in a package manager format for Windows. Be sure to mention the hash tag of the day ... #IoT

ckormanyos, 2019-12-27:
There has actually been significant progress in this area.

- `*nix` builds are supported in parallel with the VC build.
- There is also new cross-platform CMake support for the builds.

closed 2019-12-27 by ckormanyos