ckormanyos/real-time-cpp #20: cap 4 led class, toogle problem

wifixcort, 2015-12-26:
In Chapter 4 maybe have some problems 
1. In led_base class and led_port does not existe specification for type of port like existe in the second chapter 
This lines are completly missing 
<code>
*reinterpret_cast<volatile bval_type*>(port) &= static_cast<bval_type>(~bval);

*reinterpret_cast<volatile bval_type*>(port - 1U) |= bval;
</code>
When I add this lines in constructor of led_port and compile, it don't work

Then I need to add a delete function in led_port because without it I can't compile
<code>
		void operator delete(void *) {}
</code>

If the funtion not exist, this error show up

<code>
main.o: In function `led_port::~Led_port()':
main.cpp:(.text._ZN8led_portD0Ev[led_port::~led_port()]+0x0): undefined reference to `operator delete(void*)'
collect2: error: ld returned 1 exit status
</code>

ckormanyos, 2015-12-28:
This chapter and section would benefit from a sample project. I will try to create a sample project for Chapter 4, Section 1 in the next few weeks.

For missing operator delete, this is a known compiler characteristic from some compilers.

1) In the class definition, change the declaration of the destructor to the following (note pure declaration with no function body):
virtual ~led_port();

2) Below the class definition, implements the class destructor once:
class led_port
{
  // .....
};

// Implementation of the class destructor (once only!).
led_port::~led_port() { }

Please get back to me with any further questinos.
Regards, Chris.

wifixcort, 2015-12-29:
Hi, I'm currently working in that,

I attach what I did for work, and a new error I'm having with pwm.
Has a Makefile to build it simplest, with the compiler you says in the book
that one should use.
I am using the avr delay library because it is simpler to use and hex
weighs half

wifixcort, 2015-12-29:
I'm using just what it need to work to keep it simple

2015-12-29 9:05 GMT-06:00 Ricardo Mena <rgacort12@gmail.com>:

ckormanyos, 2020-04-29:
Done with example 4.4.
The description in the md file still has lots of TBDs.
Close this issue.

closed 2020-04-29 by ckormanyos