ckormanyos/real-time-cpp #22: Library with avr-ar

ajneu, 2016-02-28:
Hi Chris,

actually `mcal` would be great to place into a library archive.
I've given this a try with the code from `chapter02_03`, but think I'm doing something wrong.
Would be nice if someone could give this a try also. (perhaps someone well acquainted with gcc and/or binutils here somewhere??)

Background: I actually want to write some nice `CMakeLists.txt`, that then handle this nicely (library creation); but I need to know if I'm doing something wrong (or not?)
.
.
.

Anyway... I think I'm doing it wrong, since I get a really really small hex (I don't currently have AVR hardware, so I cannot test the hex.)

Here's what I've tried:

* **Create `bin/libmcal.a`**
 * Create the objects  
 `bin/mcal.o` `bin/mcal_gcc_cxx_completion.o` `bin/mcal_cpu.o` `bin/mcal_gpt.o` `bin/mcal_irq.o` `bin/mcal_led.o`
 * Combine those objects into an archive `bin/libmcal.a` with this command:  
 ```bash
avr-ar qc  bin/libmcal.a     bin/mcal.o bin/mcal_gcc_cxx_completion.o bin/mcal_cpu.o bin/mcal_gpt.o bin/mcal_irq.o bin/mcal_led.o
```
 * Call ranlib on the archive:
 ```bash
avr-ranlib  bin/libmcal.a
```

* **Create `bin/libstartup.a`**
 * Create the objects  
 `bin/crt0.o` `bin/crt0_init_ram.o` `bin/crt1.o` `bin/int_vect.o`
 * Combine those objects into an archive `bin/libstartup.a` with this command:  
 ```bash
avr-ar qc  bin/libstartup.a     bin/crt0.o bin/crt0_init_ram.o bin/crt1.o bin/int_vect.o
```
 * Call ranlib on the archive:
 ```bash
avr-ranlib  bin/libstartup.a
```
* **Create `bin/sys_start.o` (from `sys_start.cpp`) and link if with `bin/libmcal.a` and `bin/libstartup.a`**
 * Create object `bin/sys_start.o`
 * Link  
 ```bash
avr-g++ -x none -mrelax -nostartfiles $CFLAGS $CPPFLAGS $CINCLUDES -Wl,--gc-sections -Wl,-Ttarget/micros/avr/make/avr.ld,-Map,bin/chapter02_03.map bin/sys_start.o  -o bin/chapter02_03.elf   bin/libmcal.a bin/libstartup.a
```

ckormanyos, 2016-07-11:
I will try to look into this.

Best regards, Chris

ckormanyos, 2019-12-27:
At the present time there are 14 mcal variants for different targets. The mcal-library concept seems like it will be difficult to support for the large number of targets. Compile time is low nad the benefit of mcal libraries is expected to be low.

This issue will be closed, but the concept can be considered at a later time.

closed 2019-12-27 by ckormanyos