ckormanyos/real-time-cpp #24: Bought your book but which one is which edition?

shamun, 2017-01-06:
From Amazon, I bought your book but i am confused is it first edition i bought or second edition?
[image]

Can you please give me the URL for 1st and 2nd edition of your books?

niepiekm, 2017-01-09:
You have a 1st edition of the book. Here's a link:
https://www.springer.com/us/book/9783642429156
And here's a link to 2nd edition:
https://www.springer.com/us/book/9783662478097

shamun, 2017-01-10:
Indeed i have 1st edition, i am interested to collect your second edition too.

I am using your book specially for Chromium open source project to interconnect with real-time microcontroller. Chromium use strict C++11. 

Is your second edition completely written on strict C++11 or it mixed with 14, 17 (which i want to avoid practice only stick with C++11)?

ckormanyos, 2017-01-13:
The 1st edition is C++11 only. The 2nd edition includes uses of C++14.

The C++14 material in the 2nd edition is easy to identify. Most C++14 material is mentioned in the preface to the 2nd edition.

The companion code remains predominantly C++11.

ckormanyos, 2018-09-18:
The third edition is now available, if this is still relevant.

closed 2018-09-18 by ckormanyos