ckormanyos/real-time-cpp #25: Third edition? Not an issue :)

matejz, 2017-09-02:
Dear Mr. Kormanyos
I know that this is not the right place to ask questions, but I wasn't able to find your personal email. 
So, I'll ask here, if there are any plans for a third edition? 
Have you noticed something in the new C++ 17 standard that could warrant a new version of the book?

Thank you :)

ckormanyos, 2017-12-27:
Dear Matejz,
Yes. Sorry about the late response. New work is underway for the reference application and the examples for C++17. Thank you for your interest.
- Chris

ckormanyos, 2018-06-12:
The third edition updated for C++17 (and with more C++11 and C++14) and also with additional examples is now available. Thank you for your interest. Chris

closed 2018-09-18 by ckormanyos