ckormanyos/real-time-cpp #26: Setting up the gnu tool chain for the ATMEL ATMega328p microcontroller

Gocag, 2018-03-02:
Dear Mr. Kormanyos,
I got stuck installing the GCC prerequisites. I have never worked or built such a system. In the appendix it is not very clear on how to set them up. I am using a Windows 10 (64 bit) computer and all the prerequisites have been updated from the time that you wrote the book. 

Is there any help you can offer to put me in better position to build this tool chain.
Kind regards,
Gabriel

Gocag, 2018-03-05:
I have managed to set up the config files in each tool. i.e. PPL, GMP, binutils, mpc, mpfr, newlib and GCC. However, I can not install them using the Make and install command, because Msys can not recognise the commands. It says that they don't exist.
Any help will be highly appreciated.

ckormanyos, 2018-03-13:
Hi Gabriel. Thank you for your query. In the meantime, the dependency on PPL has been replaced by the "Integer Set Library". The full list of prerequisites can be found here [https://gcc.gnu.org/install/prerequisites.html](url)

ckormanyos, 2018-03-13:
You might need to add the path of MSYS bin to your path environment variable?

ckormanyos, 2018-03-13:
It could be easier if you search around for a prebuilt GCC for the 8-bit target. Recently I found a nice prebuilt GCC 7.3.0, but I forgot where. There are some out there. Then you do not need to spend the time building in MSYS. Good luck, Chris

ckormanyos, 2018-12-31:
As mentioned in another issue, project workspaces for Atmel(R) Studio7 have now been added to the examples. This may be of some help for this issue also.
With kind regards, Chris.

Gocag, 2019-01-07:
Hi Chris,
Thanks for the message. I'll check the examples out and work along side them.
Thank you for the follow up.
Kind regards,
Gabriel

ckormanyos, 2019-09-23:
Close this issue after adding project workspaces for Atmel(R) Studio7

closed 2019-09-23 by ckormanyos