ckormanyos/real-time-cpp #28: Building STL on Linux.

gantzm, 2018-10-15:
When running `./target/build/build.sh avr rebuild` the build fails because there is no STL implementation built.  It appears an STL implementation is provided in the `src/utils/STL` directory, but I can find no build files or instructions for building this implementation.

Is the `src/utils/STL` used, or is there a different STL dependency that needs to be downloaded and built?

gantzm, 2018-10-15:
This appears to be the issue:

The file `target/app/make/app_make.gmk` has the following:

```
C_INCLUDES = $(TGT_INCLUDES)                    \
             -I$(PATH_APP)                      \
             -I$(PATH_APP)/mcal/$(TGT)
```
While the linux build file `target/app/make/app_make_linux.gmk` has the following:

```
C_INCLUDES = -I$(PATH_APP)                      \
             -I$(PATH_APP)/mcal/$(TGT)
```

When I added the missing piece $(TGT_INCLUDES) everything appears to build fine.

ckormanyos, 2018-10-21:
Yes. I regret that the Linux build is out of date. Another yes... a small subset of the STL is implemented in the directory that you mention. Great recognition from your side.

gantzm, 2018-10-27:
Closing as the fix for this was accepted in a pull request.

closed 2018-10-27 by gantzm