ckormanyos/real-time-cpp #30: Real-Time C++ for Eclipse or Atmel Studio /.0

jureichardt, 2018-12-07:
Hi Chris,
I'm considering to use your textbook in my university course for undergrade students.
We are using Eclipse with the Arduino plugin (Sloeber V4) for C projects.  When I tried your introductionary LED-example it failed finding the <cstandard>. In this Arduino installation I also couldn't find the projects C/C++ Build option to add additional include paths or to select the g++ compiler version.
Alternatively I tried Atmel Studio 7.0 which resulted in almost the same problem.

As I can't expect my students to add all required paths, libraries and compilers manually as described in apendix C I'm asking if you have an installation script for either Atmel V7 or Eclipse/Arduino which adds all required features automatically.

With kind regards
Juergen

ckormanyos, 2018-12-30:
Project workspaces for Atmel(R) Studio7 have now been added to the examples.
Thank you for the suggestion.
With kind regards, Chris.

ckormanyos, 2019-02-22:
Project workspaces Atmel(R) Studio7 are available. Additional user test results have been obtained.
I will close this issue. Please feel free to open another issue should need arise.
Thanks again and kind regards, Chris

closed 2019-02-22 by ckormanyos