ckormanyos/real-time-cpp #32: PRs

jwinarske, 2019-06-07:
Hi Chris,

I am interested in working on adding a few platforms, and adding CMake build support.  Are you open to PRs on these areas?

Thanks,
Joel

ckormanyos, 2019-06-08:
> I am interested in working on adding a few platforms, and adding CMake build support.

Yes. Absolutely. Thank you for your interest. Looking forward to it.
Kind regards, Chris.

ckormanyos, 2019-06-09:
Just wondering... What builds to start with? Which compilers? Which compiler locations?
Good starting point could be the ref_app itself with any of the supported targets.
That is, if you elect to go for some of this work. Happy if you give it a try.
Kind regards. Chris

jwinarske, 2019-06-24:
Hi Chris.  

I just brought up the ref_app on a NXP OM13012 (LPC11C24 32k flash /8k RAM) board.  Cortex M0.  It's an old one, but a great one for CAN projects (J1939/NMEA 2000, UDS, UAV, etc).

One I clean this up, I will submit a PR.  Next goals to include Ambiq Apollo 3 Blue (384k RAM) support, and in parallel multi-host CMake support.

Joel

jwinarske, 2019-06-24:
arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]

VSCodium (w/ Cortex-Debug), and Segger JLink.

ckormanyos, 2019-06-25:
> I just brought up the ref_app on a NXP OM13093 (LPC11C24 32k flash /8k RAM) board. Cortex M0.

Fantastic! It will be good to have an M0 or M0+ running. Thank you for progressing on this topic.

This reminds me of another idea I've been considering. For ARM(R) architectures, to switch the clock base (in microseconds) to the SysTick if this has sufficiently high frequency. This would free up a timer resource and improve the "generic-ness" of the ref_app. I will look into this as well.

jwinarske, 2019-06-25:
I'm using the SysTick on [PR 33](https://github.com/ckormanyos/real-time-cpp/pull/33).  It's currently configured for 100 microseconds.

Also I created a new namespace for all the config values.  I'm using mcal::config to hold things usually layered in "project/board" layers.  If this scheme works for you, I'll use the same approach for the Apollo 3 (M4).

ckormanyos, 2019-06-25:
> I'm using the SysTick on PR 33. It's currently configured for 100 microseconds.

Yes. Cool. I get it.

> Also I created a new namespace for all the config values.
> I'm using mcal::config to hold things usually layered in
> "project/board" layers. If this scheme works for you,
> I'll use the same approach for the Apollo 3 (M4).

Fine. Please feel free to do this. I corrected what seemed like a simple cut/paste
error in your original PR of mcal_config.h in the file header guard.

ckormanyos, 2019-06-25:
I have also added the new configuration "target lpc11c24" to
the VS2017 workspace.

Great job Joel. The build worked perfectly with GCC 8.2.1.

ckormanyos, 2019-06-25:
> Next goals to include Ambiq Apollo 3 Blue (384k RAM) support,
> and in parallel multi-host CMake support

Go for it. Many thanks.

Kind regards, Chris

closed 2019-11-27 by jwinarske