ckormanyos/real-time-cpp #38: gnurx-elf build error

jwinarske, 2019-11-27:
What toolchain should I use to work around this?

**gcc_4.8.4.201902_gnurx-elf**

/media/joel/SolidState/real-time-cpp/ref_app/src/app/benchmark/app_benchmark_crc.cpp: In function ‘bool app::benchmark::run_crc()’:
/media/joel/SolidState/real-time-cpp/ref_app/src/app/benchmark/app_benchmark_crc.cpp:13:4: error: could not convert ‘{{49u, 50u, 51u, 52u, 53u, 54u, 55u, 56u, 57u}}’ from ‘<brace-enclosed initializer list>’ to ‘const mcal::memory::progmem::array<unsigned char, 9ul>’
   }};
    ^

ckormanyos, 2019-11-27:
That might be a coding problem on my side.
I will have a look.

ckormanyos, 2019-11-28:
I have reproduced this problem also with RL78 tools.
Work around, for now:
* Have switched from progmem array back to std::array.

ckormanyos, 2019-11-28:
The issue was in faulty private/deleted constructors in the special progmem array class.
These have now been corrected and aöö VC/GNUmake builds re-checked.
Result:
========== Rebuild All: 17 succeeded, 0 failed, 0 skipped ==========

closed 2019-11-28 by ckormanyos