ckormanyos/real-time-cpp #39: Latest mips-mti-elf-objcopy bug

jwinarske, 2019-11-27:
**Codescape.GNU.Tools.Package.2019.09-01.for.MIPS.MTI.Bare.Metal.CentOS-6.x86_64**

mips-mti-elf-objcopy: ref_app.hex: address 0xffffffff9d000000 out of range for Intel Hex file
mips-mti-elf-objcopy:ref_app.hex: bad value

Workarounds are either not make ihex, change it to another fomat (srec worked), or comment out step...

ckormanyos, 2019-11-27:
I have seen this before. If memory serves, this kind of problem also appeared in an ARM tool chain that was rolled out a while back. In that case, the version of objcopy or objdump had a problem (I forgot which one of the exact tools, but it was the one that extracts HEX records from ELF). Anyway...  one of those two tools had a bug when calculating a 64-bit linear address offset record in IHEX-86 language. I think all I did for workaround was simply copy objdump of objcopy from another GCC build into the bin of the existing one.
On another note, a buddy of mine is writing an ELF parser at the moment and it can extract HEX records. But that is unpublished work.
So I do not really know how to deal with this particular issue at the moment.

ckormanyos, 2019-11-27:
Oh. I see your workaround. agrees with mine.
I'll see if we can roll out an IHEX extraction tool one of these days...

closed 2019-11-28 by ckormanyos