ckormanyos/real-time-cpp #40: 32/64 bit portability

jwinarske, 2019-11-27:
Can you enable switching between -m32 and -m64?

In file included from /media/joel/SolidState/real-time-cpp/ref_app/src/os/os_cfg.h:14:0,
                 from /media/joel/SolidState/real-time-cpp/ref_app/src/os/os.h:13,
                 from /media/joel/SolidState/real-time-cpp/ref_app/src/os/os.cpp:12:
/media/joel/SolidState/real-time-cpp/ref_app/src/util/utility/util_time.h: In instantiation of ‘class util::timer<long unsigned int>’:
/media/joel/SolidState/real-time-cpp/ref_app/src/os/os_cfg.h:45:23:   required from here
/media/joel/SolidState/real-time-cpp/ref_app/src/util/utility/util_time.h:27:7: error: static assertion failed: The width of the timer tick_type can not exceed the width of mcal::gpt::value_type
       static_assert(std::numeric_limits<tick_type>::digits <= std::numeric_limits<mcal::gpt::value_type>::digits,
       ^~~~~~~~~~~~~

ckormanyos, 2019-11-27:
This is my mistake.
Actually mcal::gpt::value_type is supposed to be type defined to std::uint64_t in all MCAL configurations. Unfortunately, it looks like I forgot to do that in some of the MCAL configurations such as your new "host".
I will make mcal::gpt::value_type to be type defined to std::uint64_t in all MCAL configurations.
You can also try this and see if this config problem with static_assert is eliminated.

jwinarske, 2019-11-28:
With that change it builds with either -m32 or -m64.

I copied x86_64-w64-mingw32 to host, so it's in both.

closed 2019-11-28 by jwinarske