ckormanyos/real-time-cpp #41: Consider to use target=host for PC

ckormanyos, 2019-11-27:
Consider to use target=host for PC. At the moment, the specialized target x86_64-w64-mingw32 is used. But this does not make sense on *nix hosts.

jwinarske, 2019-11-28:
**To build master branch via Visual Studio on Windows 10 x64**

Win+R, cmd, enter

If VS variable is not set by default (default install), open Visual Studio Build Window

    cd real-time-cpp
    mkdir build
    cd build
    cmake ../ref_app -DTARGET=host
    cmake --build . --config Debug --target ALL_BUILD

**_OR_** 

Open Solution "ref_app.sln"
Right click ref_app project, and set to Default startup app
Click Debug

Following *nix pattern to build with x86_64-w64-mingw32 from MSYS or Cygwin should work too.

ckormanyos, 2019-11-28:
I just switched to:
* Use "host" when building in VC
* Treat x86_64-w64-mingw32 as any other embedded target.
I hope this change is consistent with resolving this issue?

jwinarske, 2019-11-29:
Ultimately host and x86_64-w64-mingw32 are the same.  Renaming it host was just a way to make it more generic, and provide a target for Mac, Windows, and Linux.

ckormanyos, 2019-11-30:
Understood. In the GNUmake/VC builds, the following has been done: Retain x86_64-w64-mingw32 and treat it like as one of the embedded targets. Use the new "host" for VC builds.

closed 2019-11-30 by ckormanyos