ckormanyos/real-time-cpp #44: [AppleClang] cmake ../ref_app -DTARGET=host

jwinarske, 2019-12-06:
Only issue left I see building Target host on macOS gcc/AppleClang.

```
Scanning dependencies of target ref_app
[  3%] Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark.cpp.o
[  7%] Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_complex.cpp.o
[ 11%] Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_crc.cpp.o
In file included from /Users/joel.winarske/git/real-time-cpp/ref_app/src/app/benchmark/app_benchmark_crc.cpp:5:
/Users/joel.winarske/git/real-time-cpp/ref_app/src/math/checksums/crc/crc32.h:31:71: error: no type named 'value_type' in
      'std::__1::iterator_traits<mcal::memory::progmem::forward_iterator<const unsigned char, unsigned long> >'
    using value_type = typename std::iterator_traits<input_iterator>::value_type;
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/Users/joel.winarske/git/real-time-cpp/ref_app/src/app/benchmark/app_benchmark_crc.cpp:16:27: note: in instantiation of function template specialization
      'math::checksums::crc::crc32_mpeg2<mcal::memory::progmem::forward_iterator<const unsigned char, unsigned long> >' requested here
    math::checksums::crc::crc32_mpeg2(app_benchmark_crc_data.cbegin(),
                          ^
1 error generated.
make[2]: *** [CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_crc.cpp.o] Error 1
make[1]: *** [CMakeFiles/ref_app.dir/all] Error 2
make: *** [all] Error 2

```

ckormanyos, 2019-12-06:
I will investigate.

ckormanyos, 2019-12-06:
I hope this is now fixed in master (along with some other minor corrections).

jwinarske, 2019-12-07:
Looks like a regression with the header guard:

```
In file included from /Users/joel.winarske/git/real-time-cpp_v1/ref_app/src/mcal_memory/mcal_memory_progmem_array.h:15:
/Users/joel.winarske/git/real-time-cpp_v1/ref_app/src/mcal_memory/mcal_memory_progmem_iterator.h:8:9: warning: 'MCAL_CPU_PROGMEM_ITERATOR_2019_05_04_' is used as a header guard here, followed by #define
      of a different macro [-Wheader-guard]
#ifndef MCAL_CPU_PROGMEM_ITERATOR_2019_05_04_
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/joel.winarske/git/real-time-cpp_v1/ref_app/src/mcal_memory/mcal_memory_progmem_iterator.h:9:11: note: 'MCAL_PROGMEM_ITERATOR_2019_05_04_' is defined here; did you mean
      'MCAL_CPU_PROGMEM_ITERATOR_2019_05_04_'?
  #define MCAL_PROGMEM_ITERATOR_2019_05_04_
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          MCAL_CPU_PROGMEM_ITERATOR_2019_05_04_
In file included from /Users/joel.winarske/git/real-time-cpp_v1/ref_app/src/app/benchmark/app_benchmark_crc.cpp:5:
/Users/joel.winarske/git/real-time-cpp_v1/ref_app/src/math/checksums/crc/crc32.h:31:71: error: no type named 'value_type' in 'std::__1::iterator_traits<mcal::memory::progmem::forward_iterator<const
      unsigned char, unsigned long> >'
    using value_type = typename std::iterator_traits<input_iterator>::value_type;
                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/Users/joel.winarske/git/real-time-cpp_v1/ref_app/src/app/benchmark/app_benchmark_crc.cpp:16:27: note: in instantiation of function template specialization
      'math::checksums::crc::crc32_mpeg2<mcal::memory::progmem::forward_iterator<const unsigned char, unsigned long> >' requested here
    math::checksums::crc::crc32_mpeg2(app_benchmark_crc_data.cbegin(),
                          ^

```

jwinarske, 2019-12-07:
That resolved it.

closed 2019-12-07 by jwinarske