ckormanyos/real-time-cpp #46: problems building the target with cmake in Linux

gdobato, 2019-12-31:
Hello @jwinarske , hello @ckormanyos ,

after the commit 488b7fe3 for adding support to clang, i can not build any more the target in linux(debian10). I get the error below by passing the same variables as I did when the first support for cmake was added:

gdobato@unix:~/Workspace/real-time-cpp/build_linux_host$ **cmake ../ref_app -DTRIPLE=arm-none-eabi -DTARGET=stm32f429 -DCMAKE_TOOLCHAIN_FILE=../ref_app/cmake/gcc-toolchain.cmake**
-- CMAKE_BUILD_TYPE not set, defaulting to MinSizeRel.
-- Triple ................. arm-none-eabi
-- Triple Arch ............ arm
CMake Error at cmake/gcc-toolchain.cmake:51 (get_filename_component):
  get_filename_component called with incorrect number of arguments
Call Stack (most recent call first):
  /usr/share/cmake-3.13/Modules/CMakeDetermineSystem.cmake:91 (include)
  CMakeLists.txt:33 (project)
-- Toolchain Path ......... 
-- Triple ................. arm-none-eabi
-- Triple Arch ............ arm
CMake Error at cmake/gcc-toolchain.cmake:51 (get_filename_component):
  get_filename_component called with incorrect number of arguments
Call Stack (most recent call first):
  /home/gdobato/Workspace/real-time-cpp/build_linux_host/CMakeFiles/3.13.4/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:33 (project)
-- Toolchain Path ......... 
-- The CXX compiler identification is GNU 8.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/gdobato/Workspace/MCUPlatform/target/tools/toolchain/LINUX/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /home/gdobato/Workspace/MCUPlatform/target/tools/toolchain/LINUX/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-g++
Illegal option -g
-- Check for working CXX compiler: /home/gdobato/Workspace/MCUPlatform/target/tools/toolchain/LINUX/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
Illegal option -g
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
Illegal option -g
Illegal option -g
Illegal option -g
Illegal option -g
Illegal option -g
-- Detecting CXX compile features - done
-- Generator .............. Unix Makefiles
-- Build Type ............. MinSizeRel
-- Configuring incomplete, errors occurred!
See also "/home/gdobato/Workspace/real-time-cpp/build_linux_host/CMakeFiles/CMakeOutput.log".
See also "/home/gdobato/Workspace/real-time-cpp/build_linux_host/CMakeFiles/CMakeError.log".

It seems that the variable GCC_TOOLCHAIN_PREFIX is not right set in some places. E.g, by execute_process, should it not be used as ENV variable? Before investigating further, Am I passing all necessary variables to cmake? or should I pass any additional variable to cmake?

Thanks a lot,
Gabriel

ckormanyos, 2020-01-01:
> It seems that the variable GCC_TOOLCHAIN_PREFIX is not right set in some places. E.g, by execute_process, should it not be used as ENV variable?

Thank you Gabriel. I will investigate this.

Kind regards, Chris

jwinarske, 2020-01-05:
Hi @gdobato .  Please follow this pattern, and let us know how it works.

Linux Native / Host Build
real-time-cpp$ mkdir build_linux_host && cd build_linux_host
real-time-cpp$ cmake ../ref_app -DTRIPLE=x86_64-linux-gnu -DTARGET=host -DCMAKE_TOOLCHAIN_FILE=../ref_app/cmake/gcc-toolchain.cmake
real-time-cpp$ make -j7

gdobato, 2020-01-06:
Hello @ckormanyos , hello @jwinarske,
Thanks for the answer.

Using the given pattern I get the same errors....
I investigated a little and I think I found the issue. There are 2 main problems:

**1** )In the toolchain file, in execute_process, there is an additional "-" after the variable ${GCC_TOOLCHAIN_PREFIX} (which is already set in set(GCC_TOOLCHAIN_PREFIX ${TRIPLE}**-**) and consequently, the "which" command is not able to find the path, so the BINUTILS_PATH is not set either.

By removing the "-" this issue is solved.

**2** )In the same file, the variables ${APP} and ${CMAKE_EXECUTABLE_SUFFIX} are used, but they are set in the specific target script. As a consequence, some variables (POSTBUILD_GEN_SYMBOL_LISTING, POSTBUILD_GEN_HEX, ...) are not right set, and there are errors in the postbuild phase...

As the specific target script cannot be included in gcc-toolchain.cmake since the toolchain file has no access to the cache variable ${TARGET}, this can be fixed by moving all definitions which make use of $APP and ${CMAKE_EXECUTABLE_SUFFIX} from gcc-toolchain.cmake to the root CMakeList.txt, since this one includes the specific target script.

Please, find attached a patch with the modifications. By applying it, I can build and install all output files in a Linux System, but I did not test it in Win, Apple or/and VS environment. 

[001-Fix-unix-build.txt](https://github.com/ckormanyos/real-time-cpp/files/4024286/001-Fix-unix-build.txt)

Best regards,
Gabriel

jwinarske, 2020-01-06:
Hi Gabriel,

I'll double check things build tomorrow.  What's your environment?  OS, toolchain, etc.

I did all the initial CMake work on Ubuntu 18.04.

Joel

ckormanyos, 2020-01-06:
>> I investigated a little and I think I found the issue. There are 2 main problems

> I'll double check things build tomorrow. What's your environment? OS, toolchain, etc.
> I did all the initial CMake work on Ubuntu 18.04 with all of the Toolchains.

Thanks Joel and Gabriel.

gdobato, 2020-01-06:
Hello Joel,

Ok, thank you.

- Cmake:  3.13.4
- GCC:
  For Stm32 targets: gcc-arm-none-eabi-8-2019-q3-update
  For x86_64-linux host: x86_64-linux-gnu  9.2.1 (20191008)

- OS: I have tested it in Debian 10 and Ubuntu 19.10 ( I switched to Ubuntu thinking that something in my Debian environment was wrong..., but I got the same results)

Please, note that the initial CMake support 66c2670c is also working on my side, I experienced these issues after the commit 488b7fe for clang support. In the initial support, the cache variable TRIPLE is used directly in execute_process, and the toolchain file does not use $APP and  ${CMAKE_EXECUTABLE_SUFFIX}...

Best regards,
Gabriel

jwinarske, 2020-01-06:
Thanks Gabriel.  Fix is pending in PR #48

gdobato, 2020-01-06:
Hello Chris and Joel,
Great thanks .
However the PR #48 only fixes partially the issue. The variable ${CMAKE_EXECUTABLE_SUFFIX} is not defined in gcc-toolchain.cmake, and the post-build is still broken. E.g in the case below should be elf, but it is empty:

_+message (STATUS "Executable Suffix ...  ${CMAKE_EXECUTABLE_SUFFIX}")
 set(POSTBUILD_GEN_SYMBOL_LISTING ${READELF} --syms ${APP}${CMAKE_EXECUTABLE_SUFFIX} > ${APP}_readelf.txt)
 set(POSTBUILD_GEN_HEX ${OBJCOPY} -O ihex ${APP}${CMAKE_EXECUTABLE_SUFFIX} ${APP}.hex)_

<pre><font color="#859900"><b>gdobato@unix</b></font>:<font color="#268BD2"><b>~/Workspace/real-time-cpp/build</b></font>$ cmake ../ref_app -DTRIPLE=arm-none-eabi -DTARGET=stm32f429 -DCMAKE_TOOLCHAIN_FILE=../ref_app/cmake/gcc-toolchain.cmake
-- CMAKE_BUILD_TYPE not set, defaulting to MinSizeRel.
-- Triple ................. arm-none-eabi
-- Triple Arch ............ arm
-- Toolchain Path ......... /home/gdobato/Workspace/MCUPlatform/target/tools/toolchain/LINUX/gcc-arm-none-eabi-8-2019-q3-update/bin
-- Executable Suffix ...  
-- Triple ................. arm-none-eabi
-- Triple Arch ............ arm
-- Toolchain Path ......... /home/gdobato/Workspace/MCUPlatform/target/tools/toolchain/LINUX/gcc-arm-none-eabi-8-2019-q3-update/bin
-- Executable Suffix ...  
-- The CXX compiler identification is GNU 8.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/gdobato/Workspace/MCUPlatform/target/tools/toolchain/LINUX/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /home/gdobato/Workspace/MCUPlatform/target/tools/toolchain/LINUX/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /home/gdobato/Workspace/MCUPlatform/target/tools/toolchain/LINUX/gcc-arm-none-eabi-8-2019-q3-update/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Generator .............. Unix Makefiles
-- Build Type ............. MinSizeRel
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gdobato/Workspace/real-time-cpp/build
<font color="#859900"><b>gdobato@unix</b></font>:<font color="#268BD2"><b>~/Workspace/real-time-cpp/build</b></font>$ make -j
<font color="#D33682"><b>Scanning dependencies of target ref_app</b></font>
[  2%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_crc.cpp.obj</font>
[  5%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark.cpp.obj</font>
[  8%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_fast_math.cpp.obj</font>
[ 13%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_complex.cpp.obj</font>
[ 16%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_wide_integer.cpp.obj</font>
[ 16%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_fixed_point.cpp.obj</font>
[ 18%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/led/app_led.cpp.obj</font>
[ 21%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_filter.cpp.obj</font>
[ 24%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_cpu.cpp.obj</font>
[ 27%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_float.cpp.obj</font>
[ 29%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/app/benchmark/app_benchmark_pi_spigot.cpp.obj</font>
[ 32%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_eep.cpp.obj</font>
[ 35%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_gpt.cpp.obj</font>
[ 37%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_irq.cpp.obj</font>
[ 40%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_osc.cpp.obj</font>
[ 43%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_led.cpp.obj</font>
[ 45%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_pwm.cpp.obj</font>
[ 48%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_port.cpp.obj</font>
[ 51%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_spi.cpp.obj</font>
[ 54%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/stm32f429/mcal_wdg.cpp.obj</font>
[ 56%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/os/os_task_control_block.cpp.obj</font>
[ 59%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/mcal/mcal.cpp.obj</font>
[ 62%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/os/os.cpp.obj</font>
[ 64%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/sys/idle/sys_idle.cpp.obj</font>
[ 67%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/sys/mon/sys_mon.cpp.obj</font>
[ 70%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/src/sys/start/sys_start.cpp.obj</font>
[... cut ...]
[ 97%] <font color="#859900">Building CXX object CMakeFiles/ref_app.dir/target/micros/stm32f429/startup/crt1.cpp.obj</font>
[100%] <font color="#859900"><b>Linking CXX executable ref_app.elf</b></font>
<font color="#268BD2"><b>Generating Post Build Artifacts</b></font>
readelf: Error: &apos;ref_app&apos;: No such file
make[2]: *** [CMakeFiles/ref_app.dir/build.make:615: ref_app.elf] Error 1
make[2]: *** Deleting file &apos;ref_app.elf&apos;
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/ref_app.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
</pre>

is it building on your side?
Thanks,
Gabriel

jwinarske, 2020-01-07:
PR #49 

I broke this when I refactored the post-build steps, and missed the executable suffix.  The change gives the gcc toolchain a default of .elf.  If there is ever a need for non-.elf, the post build stuff will need to move into the target include.

ckormanyos, 2020-01-07:
This should now be fixed with PR #49

closed 2020-01-08 by ckormanyos