ckormanyos/real-time-cpp #47: Explanation for Differences (between 06_01 & 01a and 17_03 & 03a)

aswimman, 2020-01-05:
Hi Chris - Back again teaching my course using your new Third Edition of the book.  Love all the updates!  Previously, you had provided an explanation of the 02_03 & 03a differences (timer blocking delay vs a tiny multi-tasking scheduler delay).  Could you provide a similar explanation for the differences for the chapter06_01 and 06_01a and the chapter17_03 & 03a?  I'm sure these were both updates since you published the third edition.  May I suggest just added a "read.me" file in the revised directories for the "a" versions.  This would help us to understand the new directories after the publishing and why you created them.  Thanks again, my students really appreciate your book!

ckormanyos, 2020-01-06:
> May I suggest just added a "read.me" file in the revised directories for the "a" versions.

Yes. I will. It is good to hear from you, Ken. Thank you for your keen insights, helpful suggestions and appreciation of the book.

Kind regards, Chris

ckormanyos, 2020-01-11:
Progress on this issue has begun, and the first md-file is available for example02_02. Progress will continue expectedly rapidly. Sorry about the delay so far.

ckormanyos, 2020-01-12:
readme files have been included up to and including 06_01a. The remaining readme files will be added soon.

aswimman, 2020-01-13:
Thanks so much Chris!

ckormanyos, 2020-01-23:
> May I suggest just added a "read.me" file in the revised directories for the "a" versions. This would help us to understand the new directories after the publishing and why you created them.

Good idea, Ken.

Readme information has bee provided for all of the examples.

If you would like further information, clarification, or if you find any errors now or in the future, please get back to me any time.

Kind regards, Chris

closed 2020-01-23 by ckormanyos