ckormanyos/real-time-cpp #51: support for ESP32

trentmillar, 2020-03-01:
It would be nice to see ESP32 based boards included within your book's code. If this is something you'd consider, this links to the toolchain pegged to GCC8.2
https://www.esp32.com/viewtopic.php?f=10&t=7400&p=31257#p31257

ckormanyos, 2020-03-01:
> It would be nice to see ESP32 based boards included within your book's code.

Thank you for this idea, which I appreciate very much. Over the years, there have been a few comments regarding added targets.

At the moment, the reference application supports numerous targets in addition to the one in the book. There is a start on supporting the ESP32 (xtensa32).

But the code still needs significant work on the [not-yet-finished mcal layer for ESP32](https://github.com/ckormanyos/real-time-cpp/tree/master/ref_app/src/mcal)

Up to this point, all supported targets use exclusively hand-written code very close to the metal. And I have not yet extracted startup code, clock initialization and a timer base for this architecture yet at the moment.

I will keep posted on any upcoming progesss as it arises.

Kind regards, Chris

ckormanyos, 2020-03-03:
> Up to this point, all supported targets use exclusively hand-written code very close to the metal.

For this target, I am considering a different approach of taking the existing firmware and building the little MCAL on top of the existing stuff in a free RTOS task.

ckormanyos, 2020-07-15:
@trentmillar

I was wondering if [this](../../../../espressif/ESP8266_RTOS_SDK) is a good starting point for building a sekeleton free RTOS task based on a skinny bare metal firmware.

Given a bit of time and a few tutorials, I can probably figure out the bootloader from there.

trentmillar, 2020-08-27:
This is a great starting point @ckormanyos. I know this would be helpful for myself but I'm sure a lot of others too.

ckormanyos, 2020-08-28:
> This is a great starting point

Thank you @trentmillar

I am, in fact, slowly but surely working through the issues and the requests for new targets

And ESP32 is high on the list (just after I re-activate BeagleBone Black). I just finished RPI Zero bare metal startup.

ckormanyos, 2020-12-26:
- Work has begun in a branch.
- Build and blink OK. Starting point was Arduino Studio blinky. Not ready for PR yet.
- There are lots of program artifacts in the underlying SDK and we will try to remove the unused ones.

ckormanyos, 2020-12-27:
Hi @trentmillar there is finally preliminary support for ESP32.
I do not know if there is still interest in this rather old issue.
Anyway, I decided to place the standard multitasking scheduler on top of a slimmed down variation of the Arduino Framework.
The microsecond time is extracted from a standard function in the ESP32 SDK, as are port manipulations.

PR #71 (2020-12-27): Esp32 bare metal
Fixes #51 preliminary support for ESP32. Use multitasking scheduler on top of slimmed down Arduino Framework
closed 2020-12-27 by ckormanyos