ckormanyos/real-time-cpp #54: Sometimes the argument in UINT8_C is a suffixed integer but it should be an unsuffixed integer

fivdi, 2020-06-12:
In [this code](https://github.com/ckormanyos/real-time-cpp/blob/master/ref_app/target/micros/bcm2835_raspi_b/startup/int_vect.cpp#L74-L84):
```cpp
    static const std::uint8_t the_lower_interrupt_vector_table_data[64U] =
    {
      UINT8_C(0x18U), UINT8_C(0xF0U), UINT8_C(0x9FU), UINT8_C(0xE5U), UINT8_C(0x18U), UINT8_C(0xF0U), UINT8_C(0x9FU), UINT8_C(0xE5U),
      UINT8_C(0x18U), UINT8_C(0xF0U), UINT8_C(0x9FU), UINT8_C(0xE5U), UINT8_C(0x18U), UINT8_C(0xF0U), UINT8_C(0x9FU), UINT8_C(0xE5U),
      UINT8_C(0x18U), UINT8_C(0xF0U), UINT8_C(0x9FU), UINT8_C(0xE5U), UINT8_C(0x18U), UINT8_C(0xF0U), UINT8_C(0x9FU), UINT8_C(0xE5U),
      UINT8_C(0x18U), UINT8_C(0xF0U), UINT8_C(0x9FU), UINT8_C(0xE5U), UINT8_C(0x18U), UINT8_C(0xF0U), UINT8_C(0x9FU), UINT8_C(0xE5U),
      UINT8_C(0x00U), UINT8_C(0x84U), UINT8_C(0x00U), UINT8_C(0x00U), UINT8_C(0x40U), UINT8_C(0x80U), UINT8_C(0x00U), UINT8_C(0x00U),
      UINT8_C(0x80U), UINT8_C(0x80U), UINT8_C(0x00U), UINT8_C(0x00U), UINT8_C(0xC0U), UINT8_C(0x80U), UINT8_C(0x00U), UINT8_C(0x00U),
      UINT8_C(0x00U), UINT8_C(0x81U), UINT8_C(0x00U), UINT8_C(0x00U), UINT8_C(0x40U), UINT8_C(0x81U), UINT8_C(0x00U), UINT8_C(0x00U),
      UINT8_C(0x80U), UINT8_C(0x81U), UINT8_C(0x00U), UINT8_C(0x00U), UINT8_C(0xC0U), UINT8_C(0x81U), UINT8_C(0x00U), UINT8_C(0x00U),
    };
```
the argument in `UINT8_C` is always a suffixed integer as the suffix 'U' is always used. However, according to [7.20.4 Macros for integer constants](http://port70.net/~nsz/c/c11/n1570.html#7.20.4)

> The argument in any instance of these macros shall be an unsuffixed integer constant (as defined in 6.4.4.1) with a value that does not exceed the limits for the corresponding type.

Depending on how `UINT8_C` is defined, the compiler may or may not detect this error.

If `UINT8_C` is defined [like this](https://github.com/ckormanyos/real-time-cpp/blob/master/ref_app/src/util/STL/cstdint#L32):
```cpp
#define UINT8_C(x) ((uint8_t) (x##U))
```
the compiler will display an error something like this:
```
main.cxx:16:27: error: unable to find numeric literal operator 'operator""UU'
   16 | std::uint8_t ui = UINT8_C(0xffU);
      |                           ^~~~~
main.cxx:15:32: note: in definition of macro 'UINT8_C'
   15 | #define UINT8_C(x) ((uint8_t) (x##U))
      |                                ^
main.cxx:16:27: note: use '-fext-numeric-literals' to enable more built-in suffixes
   16 | std::uint8_t ui = UINT8_C(0xffU);
      |                           ^~~~~
main.cxx:15:32: note: in definition of macro 'UINT8_C'
   15 | #define UINT8_C(x) ((uint8_t) (x##U))
```

If `UINT8_C` like this:
```cpp
#define UINT8_C(x)      x
```
the compiler will not display an error.

ckormanyos, 2020-06-12:
Thank you. Very good find!

ckormanyos, 2020-06-12:
If you happen to find any other similar cases, please report them.

closed 2020-06-12 by ckormanyos

fivdi, 2020-06-12:
> If you happen to find any other similar cases, please report them.

Will do. To the best of my knowledge there are no similar cases or at least I haven't been able to find any with `grep`.

commit 050bc89a (2020-06-12): Fixes #54