ckormanyos/real-time-cpp #55: Execute permission for bash scripts not set on Linux

fivdi, 2020-06-13:
If the repository is cloned on Linux with the command `git clone https://github.com/ckormanyos/real-time-cpp.git` the execute permission bit for each of the bash build scripts is not set.

For example, as can be seen in the directory listing for chapter04_04 below, the execute permission bit for `build.sh` is not set.
```
$ ls -la
total 144
drwxrwxr-x  5 brian brian  4096 Jun 12 23:39 .
drwxrwxr-x 16 brian brian  4096 Jun 12 23:39 ..
-rw-rw-r--  1 brian brian  5966 Jun 12 23:39 build.bat
-rw-rw-r--  1 brian brian  6498 Jun 12 23:39 build.sh
-rw-rw-r--  1 brian brian   949 Jun 12 23:39 chapter04_04.atsln
-rw-rw-r--  1 brian brian 21897 Jun 12 23:39 chapter04_04.cppproj
-rw-rw-r--  1 brian brian  1624 Jun 12 23:39 chapter04_04.sln
-rw-rw-r--  1 brian brian 33029 Jun 12 23:39 chapter04_04.vcxproj
-rw-rw-r--  1 brian brian 18877 Jun 12 23:39 chapter04_04.vcxproj.filters
drwxrwxr-x  2 brian brian  4096 Jun 12 23:39 images
-rw-rw-r--  1 brian brian  3604 Jun 12 23:39 readme.md
drwxrwxr-x 11 brian brian  4096 Jun 12 23:39 src
drwxrwxr-x  4 brian brian  4096 Jun 12 23:39 target
-rw-rw-r--  1 brian brian 11813 Jun 12 23:39 target.vcxproj
-rw-rw-r--  1 brian brian  2315 Jun 12 23:39 target.vcxproj.filters
$ 
```
Attempting to run the command `./build.sh` results in an error:
```
$ ./build.sh
bash: ./build.sh: Permission denied
$
```
It's possible to resolve this issue by setting the execute permission bit for `build.sh` with the command `chmod +x build.sh`. However, it would be nice if the execute permission bits for bash scripts were already set in the GitHub repository so that this manual intervention could be avoided.

ckormanyos, 2020-06-14:
This will be handled both here i nthis issue as well as in #56

mark-1003, 2020-06-14:
Hello @ckormanyos,

is there a way to create a Pull-Request to this project?
I already created a bugfix branch with the fix of the missing execute permissions for bash scripts...

BR,
Mark

ckormanyos, 2020-06-14:
> is there a way to create a Pull-Request to this project?

Yes. Of course. Just create a normal PR for your branch. I can integrate the PR immediately thereafter.

Thank you and kind regards, Chris.

mark-1003, 2020-06-14:
Should be solved for all shell scripts in Pull-Request #59 
:)

fivdi, 2020-06-14:
Fixed by #59

closed 2020-06-14 by fivdi