ckormanyos/real-time-cpp #56: Ensure that the bash shells for examples build

ckormanyos, 2020-06-14:
Over time some of the bash files for the examples may have become broken or were migrated improperly. 

1) Ensure that the bash files for the examples all function properly
2) Are the file lists correct?
3) Set execution rights properly with `chmod`.

fivdi, 2020-06-14:
An additional potential improvement would be to add a bash build script for [examples/chapter03_02](https://github.com/ckormanyos/real-time-cpp/tree/master/examples/chapter03_02) as it currently doesn't have one.

ckormanyos, 2020-06-14:
> An additional potential improvement would be to add a bash build script for examples/chapter03_02 as it currently doesn't have one.

Good catch. Thank you. Will do.

ckormanyos, 2020-06-17:
The bash shells are out of synchronization for some examples.
The main problems are:

- Files not properly listed.
- Name of ELF/HEX, etc. not in line with example name.

For Win* batch files, see #60

ckormanyos, 2020-06-20:
Progress has slowly begun.
Example chapter03_02 bash shell and DOS batch builds are fixed.

fivdi, 2020-06-20:
Nice. I can now compile and flash the example for chapter03_02. It was necessary to manually set the execution bit on `build.sh` as it's not set in the repository. I don't actually know if the example is running correctly as I'm not 100% sure what visible feedback should be seen when the example runs. I'll have to take a closer look at the code.

One small point is that the first line displayed by `./build.sh` is:
```
Building with        : build.bat
```
but it should be:
```
Building with        : build.sh
```

```
brian@linuxbox:~/src/real-time-cpp/examples/chapter03_02$ ./build.sh
bash: ./build.sh: Permission denied
brian@linuxbox:~/src/real-time-cpp/examples/chapter03_02$ chmod +x build.sh 
brian@linuxbox:~/src/real-time-cpp/examples/chapter03_02$ ./build.sh

Building with        : build.bat
Using tool path      : /home/brian/opt/arduino-1.8.10/hardware/tools/avr/bin
Using tool prefix    : avr
Remove bin directory : bin/
Create bin directory : bin/

Compile  : app_led.cpp to bin/app_led.o
Compile  : app_prime.cpp to bin/app_prime.o
Compile  : mcal.cpp to bin/mcal.o
Compile  : mcal_gcc_cxx_completion.cpp to bin/mcal_gcc_cxx_completion.o
Compile  : mcal_cpu.cpp to bin/mcal_cpu.o
Compile  : mcal_eep.cpp to bin/mcal_eep.o
Compile  : mcal_gpt.cpp to bin/mcal_gpt.o
Compile  : mcal_irq.cpp to bin/mcal_irq.o
Compile  : mcal_led.cpp to bin/mcal_led.o
Compile  : mcal_osc.cpp to bin/mcal_osc.o
Compile  : mcal_port.cpp to bin/mcal_port.o
Compile  : mcal_pwm.cpp to bin/mcal_pwm.o
Compile  : mcal_spi.cpp to bin/mcal_spi.o
Compile  : mcal_wdg.cpp to bin/mcal_wdg.o
Compile  : os.cpp to bin/os.o
Compile  : os.cpp to bin/os_task_control_block.o.
Compile  : sys_idle.cpp to bin/sys_idle.o
Compile  : sys_mon.cpp to bin/sys_mon.o
Compile  : sys_start.cpp to bin/sys_start.o
Compile  : crt0.cpp to bin/crt0.o
Compile  : crt0_init_ram.cpp to bin/crt0_init_ram.o
Compile  : crt1.cpp to bin/crt1.o
Compile  : int_vect.cpp to bin/int_vect.o
Link     : objects to bin/chapter03_02.elf

Extract  : executable hex file : from bin/chapter03_02.elf
Extract  : assembly list file  : from bin/chapter03_02.elf
Extract  : size information    : from bin/chapter03_02.elf
Extract  : name information    : from bin/chapter03_02.elf
Extract  : demangled names     : from bin/chapter03_02.elf
brian@linuxbox:~/src/real-time-cpp/examples/chapter03_02$ sudo $(which avrdude) -C $AVRDUDE_CONFIG -c dragon_isp -P usb -p m328p -e -U flash:w:./bin/chapter03_02.hex 

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: erasing chip
avrdude: reading input file "./bin/chapter03_02.hex"
avrdude: input file ./bin/chapter03_02.hex auto detected as Intel Hex
avrdude: writing flash (6198 bytes):

Writing | ################################################## | 100% 6.67s

avrdude: 6198 bytes of flash written
avrdude: verifying flash memory against ./bin/chapter03_02.hex:
avrdude: load data flash data from input file ./bin/chapter03_02.hex:
avrdude: input file ./bin/chapter03_02.hex auto detected as Intel Hex
avrdude: input file ./bin/chapter03_02.hex contains 6198 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 6.68s

avrdude: verifying ...
avrdude: 6198 bytes of flash verified

avrdude: safemode: Fuses OK (E:FF, H:D9, L:62)

avrdude done.  Thank you.

brian@linuxbox:~/src/real-time-cpp/examples/chapter03_02$ 
```

ckormanyos, 2020-06-21:
> but it should be: `Building with        : build.sh`

Good catch. Thank you.
This has been corrected.

> not 100% sure what visible feedback should be seen when the example runs

This particular example does not offer very mush visible verification
other than the 1/2Hz blinky reflecting proper error-free execution
and check of the prime sieve calculation.

There is, however, a debug port toggle built in for a runtime scope measurements

A clear description of this feature and a few details about what to expect
when running 03_02 are provided in the bottom of the page in the extended
[description of example chapter03_02](https://github.com/ckormanyos/real-time-cpp/tree/master/examples/chapter03_02)

ckormanyos, 2020-06-21:
Please feel free to provide indication, either positive or negative, or other
suggestions regarding your experience with 03_02.

Play around with the compiler optimization flags to investigate runtime per size
optimization potential once you have the cycle task runtime on the scope.

fivdi, 2020-06-21:
> Please feel free to provide indication, either positive or negative, or other
suggestions regarding your experience with 03_02.

Up until now I have only read the first three chapters of the book. To a certain extent, as a reader, I felt a little like I was been throw into the deep end with  examples 02_03, 02_03a and 03_02. At least this is how I felt initially. This feeling was stronger with example 03_02. If I'm not mistaken, these examples use quite a bit of code that isn't discussed till later in the book. On the other hand, I don't expect to be spoon-fed so I was able to figure out what I needed to know. Figuring things out was relatively easy for 02_03, 02_03a. For 03_02 it was more time consuming. Today, a new section was added to the readme; [Running the Setup](https://github.com/ckormanyos/real-time-cpp/tree/master/examples/chapter03_02#running-the-setup). This improves things a lot. Yesterday evening and earlier this morning this section didn't exit. The presence of the file [mcal_port_expander_microchip_mcp23s17.h](https://github.com/ckormanyos/real-time-cpp/blob/master/examples/chapter03_02/src/mcal/avr/mcal_port_expander_microchip_mcp23s17.h) led me to believe that an MCP23S17 I/O port was being used. In addition, as least for me, the presence of code related to SPI in the assembly listing file was further confirmation that SPI was being used. It took me a while to figure out that this isn't the case.

I'm not using an IDE to look at the code. Perhaps I should be. If an IDE is not being used a few searches with `grep` or similar are needed to figure out that port d bit 3 is used by [this line of code](https://github.com/ckormanyos/real-time-cpp/blob/master/examples/chapter03_02/src/app/prime/app_prime.cpp#L44):
```
app_prime_port_type::set_direction_output();
```
I understand why it's called `app_prime_port_type` and not `app_prime_portd_b3_type` but nevertheless the searching is tedious.

I tend to look at the assembly code in the listing file quite a bit. However, the listing file contains mangled symbol names which makes it difficult to read. As an example, here's the assembly code for `main()` in `03_02` with mangled names:
```
000012d8 <main>:
    12d8:	a0 d8       	rcall	.-3776   	; 0x41a <_ZN4mcal4initEv>
    12da:	62 dd       	rcall	.-1340   	; 0xda0 <_ZN2os8start_osEv>
    12dc:	90 e0       	ldi	r25, 0x00	; 0
    12de:	80 e0       	ldi	r24, 0x00	; 0
    12e0:	08 95       	ret
```
To demangle symbol names I edit `build.sh` and pass the `-C` option to `avr-objdump` and get the following assembly code for `main()` with demangled names which is easier to read:
```
000012d8 <main>:
    12d8:	a0 d8       	rcall	.-3776   	; 0x41a <mcal::init()>
    12da:	62 dd       	rcall	.-1340   	; 0xda0 <os::start_os()>
    12dc:	90 e0       	ldi	r25, 0x00	; 0
    12de:	80 e0       	ldi	r24, 0x00	; 0
    12e0:	08 95       	ret
```
Perhaps you could use the `-C` option when generating assembly listings.

> Play around with the compiler optimization flags to investigate runtime per size
optimization potential once you have the cycle task runtime on the scope.

After reading this I noticed that `build.sh` isn't using the options `-fdata-sections -ffunction-sections`. Adding both of these options to `CFLAGS` will decrease the size of the text section for 03_02 from 5888 to 3632 bytes. This means that the text segment is currently approximately 62% larger than it needs to be.

In the assembly listing file there are 68 bytes of SPI specific code in the address range 0x7ac to 0x7ef. I don't think it's actually needed for anything.

I hope all of the above doesn't sound too negative as this isn't my intention. To be honest, in retrospect I think it's better to have been thrown into the deep end with 03_02 rather than to have been spoon-fed.

ckormanyos, 2020-06-21:
Thank you for providing your excellent and detailed feedback.

> into the deep end with examples 02_03, 02_03a and 03_02

That was the intent, in an ultr-practical, almost irreverent way. I will, however look into ways to provide more clear steps along the way. Your identification of unused files and even chip drivers in 03_02 have motivated me to reduce those kinds of things and make sure that the examples are lean and clear as possible.

> The presence of the file mcal_port_expander_microchip_mcp23s17.h led me to believe that an MCP23S17 I/O port was being used. In addition, as least for me, the presence of code related to SPI in the [...]

This is simply an overlooked mistake. The SPI and the external chip driver do not belong in this example. I am in the process of cleaning up all of the examples. And this cleanup is still underway. The misplaced files will be removed from examples where not needed.

> isn't using the options `-fdata-sections` `-ffunction-sections`

These are intended to be used. A historical mistake seems to have removed them from bash and batch. I will re-insert them.

> Perhaps you could use the `-C` option when generating assembly listings

I will investigate this. Thanks. I also should produce a symbol listing of unmangled names with `c++filt`. I will improve the situation for the study of the listings and maps.

Thanks again and kind regards, Chris

ckormanyos, 2020-06-27:
fixed by #62

closed 2020-06-27 by ckormanyos