ckormanyos/real-time-cpp #96: Benchmark tests to run on simulated ARM via QEMU in CI

ckormanyos, 2021-01-24:
In further support of CI measures, we will make a first step by getrting the benchmark tests to run on a simulated ARM target such as ARM-v8 or Cortex via QEMU in CI.

This issue has bee partly motivated by further discussion in #45, originally motivated in #37

ckormanyos, 2021-01-24:
Recal this comment from the OP in #45...

> covers most everything on using QEMU for aarch64 bare metal. ARMv7, and QEMU AVR would follow the same pattern. CTest could be used to execute the unit tests; minor CMake changes.

ckormanyos, 2021-03-22:
Made a big step forward here with 4052c17fb2f3457273440c271119fc04c17f1bc8, having some Boost.Math stuff building successfully _on the metal_ for `target=stm32f446`. Will pursue this issue here for the potential purpose of getting similar CI (if desired) running for Boost.

ckormanyos, 2021-03-25:
Got a [single-file benchmark](https://github.com/ckormanyos/real-time-cpp/runs/2197479767?check_suite_focus=true) running in CI with standalone LD flags, own startup etc.

Hi @gdobato this should be ready for initial prototyping on QEMU.

gdobato, 2021-03-27:
Hello @ckormanyos ,
I added the benchmark-single for stm32f429 since it is already supported by arm-qemu (stm32f4446 it is not supported yet, although in essence it should not be a problem since benchmark does not used any peripheral at all, and target memory layout looks identical). #105 can be merged to avoid conflicts in .yml

I need to work in some local tests before going forward with qemu in CI.

BR,

gdobato, 2021-04-03:
Hello @ckormanyos ,
I did some progress... please check #106 out. Remarks:
1.- qemu already runs in CI. 
2.- gdb-py is not included in the gcc-arm toolchain available in default ubuntu repo (apt "ed"), so i use for the moment wget to get it from  https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads, which does include gdb-py
3.-used qemu variant requires npm, xpm package managers to install it, so i decided to also use wget for moment too...
4.- Just one simple test is running, which sets a break point in main, and continue execution... not real benchmark result is checked. ref_app/target/build/test_app_benchmarks_emulator.py still needs some work..., still is a big step.
Best regards,
Gabriel

ckormanyos, 2021-04-03:
> did some progress...

Thank you very much, Gabriel. That is fascinating technology and great progress.
#106 has been merged into main.

The line where a modern GCC arm-non-eabi is obtained as a tar ball is really cool!

I really appreciate your work, all of it.

ckormanyos, 2022-02-26:
Note to self: Take note of [this](https://github.com/nasa-jpl/embedded-gcov), which seems like it could help extracting coverage info as well. This could be cool.

ckormanyos, 2022-07-02:
Hi Gabriel (@gdobato) I finally got back to this issue. Based on your work, there are some extensions cycling in CI now. It is now possible to successfully assess the benchmark result and use this information within the calling LINUX shell on CI.
- I ended up modifying the code a bit in order to create a secondary subroutine running after a primary subroutine
- The primary subroutine performs the benchmark and latches in a magic value success constant (I'm using `0xF00DCAFE` to mark success.
- A breakpoint is set in the secondary subroutine which latches in the successful value and passes this to main.
- The successful magic flag is printed via the Pythin GDB interface to the screen. It's decimal value is used.
- The entire GDB/Python-controlled session is piped to a text file, which is subsequently printed with `cat` and `sed`-piped-to-`grep`-ed for the magic constant value latched in the breakpoint indicating success.
- Success of `grep` is used by Ubuntu Shell to assess success.

This whole thing could be a lot more refined if the Python script were to return success/fail back to the shell. We can do this in a second round according to time/interest.

Anyway it works. The tests acr cycling in #267 

Cc: @imahjoub you will be interested in the QEMU run on Ubuntu in Ci running on a bare metal CM4 Core. Cool!

commit adb076fc (2022-07-02): Merge pull request #267 from ckormanyos/more_qemu_progress

Fix #96 via QEMU refactor and extension
closed 2022-07-02 by ckormanyos