ckormanyos/real-time-cpp #101: am335x not building on cmake

ckormanyos, 2021-02-05:
For some reason, target am335x is not building on cmake. It appears to be the link stage, although at first glance it's hard to see how this differs from the successful build on GNUmake.

ckormanyos, 2022-02-25:
Indication that the linker for some reason can't find `_exit()`. Possibly related to #231

ArshaShiri, 2022-02-26:
Dear @ckormanyos, 
I was curious about this issue so I looked a bit into it. I noticed in the am335x.cmake file on line 33 `set(CMAKE_EXECUTABLE_SUFFIX "")` the suffix is not set to `.elf`. Is that intentional? By setting the suffix to .elf, I was able to build this on my system.

ckormanyos, 2022-02-26:
> I noticed in the am335x.cmake file on line 33 `set(CMAKE_EXECUTABLE_SUFFIX "")` the suffix is not set to `.elf`.

Thank you. Yes. That solves the problem. Fantastic catch.

> Is that intentional?

No. That seems like a mistake that slipped in, perhaps due to the fact that the BBB needs a kernel binary image to run from micro-SD card. But that needs to be extracted from ELF anyway. So the missing `.elf` suffix is simply a mistake.

ArshaShiri, 2022-02-26:
@ckormanyos  I'm very glad that issue is fixed. I just tried to publish a branch with the fix but I am missing the permission to do so. But also I saw that a PR is already created for it :-)

ckormanyos, 2022-02-26:
Thanks @ArshaShiri it's great that you fixed a tough bug issue (and the issue is more than 1 year old)!

Sorry if I went too quickly on the PR. Anyway the fix is cycling on CI and should get into the code base today or tomorrow.

> publish a branch with the fix but I am missing the permission to do so

Hmmm... I admit those permission settings are something I'm a bit confused on. We have had contirubutors previously. Did you fork and commit to fork? I think you might need to fork the repo, make a branch then submit a PR from your branch on the fork back to my main branch. i usually use third party tools for this kind of thing, so I'm not sure exactly how the commands all work out.

ArshaShiri, 2022-02-26:
@ckormanyos No problem at all. I cloned the repo and created a branch and tried to publish that. That's why it did not work probably. Thank your very much for the explanation!

commit 600f954b (2022-02-26): Fix #101 am335x build on CMake
closed 2022-02-26 by ckormanyos