ckormanyos/real-time-cpp #231: Not able to build via CMake for stm32f407

ArshaShiri, 2022-02-23:
When running 

> cmake ../ref_app -DCROSS_COMPILE=arm-none-eabi -DTARGET=stm32f407 -DCMAKE_TOOLCHAIN_FILE=../ref_app/cmake/stm32f407.cmake

it seems that the CXX compilers is rolled back to GNU on my system:

> -- CMAKE_BUILD_TYPE not set, defaulting to MinSizeRel.
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/c++
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Generator .............. Unix Makefiles

I explicitly set the compiler to arm-none-eabi-gcc in stm32f407.cmake and that led to `undefined reference to `_exit'` error in CMake. I was wondering if this is an issue or I am missing something. 
Thank you very much for your time and help.

ckormanyos, 2022-02-25:
> explicitly set the compiler to `arm-none-eabi-gcc` in stm32f407.cmake and that led to undefined reference to `_exit'` error in CMake. I was wondering if this is an issue or I am missing something.

Hi @ArshaShiri thanks for your report. Sorry about the late response on this issue.

Actually, I believe there is an issue here. When compiling with CMake for `am335x`, we have also observed this phenomenon. It is being tracked in #101 (now linked to this issue).

I've tried a few things. One thing we need to make sure of is if we are linking in `CMake` with the so-called `nano-specs`, as hat compiler might be built with newlib-nano. I need to check if I incorporated that into the call of the linker in the CMake build scripts.

I wonder if you could try to see if you can build with GNUmake? If so, then that tells us if the problem is CMake-dependencies-related or generally on the distro that you've spinned up.

This issue has perplexed me for a while and a few people (non-published) have looked into this without yet finding a solution. So i am very keen on making progress on this issue and hearing about any ideas or experiences you make...

ckormanyos, 2022-02-25:
I actually label this issue as a bug, since I think it is a bug, though I'm not exactly sure where or which kind of bug we are dealing with.

ckormanyos, 2022-02-25:
Hi @ArshaShiri also I notice the docs are poor in this case.

The pattern for building with CMake for stm32f407 is:

```sh
cd real-time-cpp
mkdir build
cd build
cmake ../ref_app -DTRIPLE=arm-none-eabi -DTARGET=stm32f407 -DCMAKE_TOOLCHAIN_FILE=../ref_app/cmake/gcc-toolchain.cmake
make -j ref_app
```

I will correct the docs.

But this might still not fully solve the issue you mention.

ArshaShiri, 2022-02-25:
```
cd real-time-cpp
mkdir build
cd build
cmake ../ref_app -DTRIPLE=arm-none-eabi -DTARGET=stm32f407 -DCMAKE_TOOLCHAIN_FILE=../ref_app/cmake/gcc-toolchain.cmake
make -j ref_app
```

closed 2022-02-25 by ArshaShiri

ArshaShiri, 2022-02-25:
Dear @ckormanyos 
Thank you very much for your reply and help. Using your suggested pattern the issue is resolved. 
The problem was that I was setting the CMAKE_TOOLCHAIN_FILE to stm32f407.cmake instead of gcc-toolchain.cmake which was my mistake. Sorry for taking your time for this.
I can also build using GNUMake by the way, so there seems to be no problem for stm32f407. I tired building the am335x and no elf was generated indeed when I used CMake.

ckormanyos, 2022-02-25:
Thank you @ArshaShiri

> problem was that I was setting the CMAKE_TOOLCHAIN_FILE

I am glad it works locally for you now. Your very clear report helped point out weaknesses in the docs of `real-time-cpp`. Thank you for that.

> can also build using GNUMake

Excellent! Thanks for checking.

If any other problems or roadblocks are encountered, please feel free to query any time.

> tired building the am335x and no elf was generated indeed when I used CMake

Yes. that is another independent confirmation of what seems to be the topic of #101, which will remain open.