ckormanyos/real-time-cpp #268: Modernize the QEMU run(s) and include CNL benchmark

ckormanyos, 2022-07-08:


closed 2022-07-08 by ckormanyos

ckormanyos, 2022-07-12:
Hi John (@johnmcfarlane) for your information, in this closed thread...

I finally got around to finishing a draft of QEMU for bare-metal STM32F429 (Cortex-M4, no floating-point unit emulation) running a tiny CNL benchmark on the (simulated) metal in CI. The benchmark features a straightforward use of [CNL's scaled integer](https://github.com/ckormanyos/real-time-cpp/blob/master/ref_app/src/app/benchmark/app_benchmark_cnl_scaled_integer.cpp).

I run GDB via python script and capture the value of a debug variable in the GDB session (0xF00DCAFE) having decimal value 4027435774, which is subsequently `grep`-ed out of the piped GDB session log. It's kind of a non-sophisticated approach which can and probably will be defined. But it's kind of cool. My CI job clones CNL's main branch, compiles the benchmark with `arm-none-eabi` using a custom startup code and linker script and subsequently runs the benchmark.

The exact CI run is [here](https://github.com/ckormanyos/real-time-cpp/blob/242a5de9500272c5901a83cf951869e5eb5454e0/.github/workflows/real-time-cpp.yml#L830).

Thanks for providing CNL.

johnmcfarlane, 2022-07-12:
Hi @ckormanyos. Thanks for the update. That's great news and great work! (I've [not forgotten](https://github.com/ckormanyos/wide-integer/issues/183#issuecomment-1013875637) that this is a valuable application of CNL.)

commit 242a5de9 (2022-07-08): Merge pull request #269 from ckormanyos/further_qemu_progress

Fix #268 via modernize qemu CI runs