ckormanyos/real-time-cpp #270: Difficulty in starting work

sami1369, 2022-07-15:
The book and git don't have a proper tutorial to start with and the content is very confusing.
 
Please help me to upload the code in the right way

Regards

ckormanyos, 2022-07-16:
> book and git don't have a proper tutorial to start with and the content is very confusing.

> help me to upload the code in the right way

Thanks for your query @sami1369. Let's start with some basics. You can obtain the code by simply clicking on the "code" button and downloading the ZIP archive. See the picture below.

[image]

ckormanyos, 2022-07-16:
Once you have downloaded the project, we can build one of the samples or the reference application. i can help you along those lines.

What Operating System are you using?

sami1369, 2022-07-16:
Hi Christopher,

I searched for your email, but unfortunately, I was not successful. I had
to send a message here to you.

My hardware is Arduino UNO, and I don't know what I should do to compile
and upload it on the board?

Kind regards,
Saeid

sami1369, 2022-07-16:
> Once you have downloaded the project, we can build one of the samples or the reference application. I can help you along those lines.
> 
> What Operating System are you using?

Yes, I downloaded it and put it in a folder where Mingw-w64 is there

ckormanyos, 2022-07-17:
> I downloaded it and put it in a folder where Mingw-w64 is there

OK. Good.

There are several ways to build and use the companion code. Builds are supported on LINUX, Windows, using GNUmake or CMake. But you are right, the build possibilities can, in fact, be confusing when starting.

A lot of folks have used LINUX systems. Others use Windows. A build system specifically targeting MinGW is not explicitly supported but probably would work with a bit of effort.

It can be easiest to first build for the Windows/LINUX host (in a command shell). As a second step, you can go for an embedded target build.

Of the following choices, how would you like to start?
1. Build in MICROCHIP's ATMEl Studio.
2. Build on LINUX with GNUmake.
3. Build on LINUX with CMake.
4. Build on Windows in Microsoft Visual Studio (using specially ported compilers [here](https://github.com/ckormanyos/real-time-cpp-toolchains)).
5. Standalone build on the Windows command line.

Believe it or not, choice 5 can be extremely straightforward, requiring no added installations, just getting the toolchain, unpacking, moving it and executing the build command line. These exact instructions are shown [here](https://github.com/ckormanyos/real-time-cpp/blob/d006c98f3c433124bc5b057dc36215d50723fa3b/.github/workflows/real-time-cpp.yml#L558). But instead of building with MSBuild (as shown on [line 589](https://github.com/ckormanyos/real-time-cpp/blob/d006c98f3c433124bc5b057dc36215d50723fa3b/.github/workflows/real-time-cpp.yml#L589), a simple command line could be used.

ckormanyos, 2022-07-17:
> Believe it or not, choice 5 can be extremely straightforward

If you diecde to go this way, that can be simple. When you've unpacked the toolchains, I con provide you with help regarding setting up the command line to build for Windows host.

sami1369, 2022-07-18:
I downloaded it twice, but the file has a problem and cannot be unpacked. Since it has a large volume, let me download and unpack it again; maybe the problem will be solved.

sami1369, 2022-07-20:
After After downloading a few times I get error when I want to unzip it: 
[image]
can you give me another link to download?

sami1369, 2022-07-20:
> I just downloaded the real-time-cpp-toolchains repository as a ZIP
> archive from GitHub and unpacked it with 7-zip. For me it works fine.
>
> I'm not sure what your favorite unpacker is, but I've had good results
> with 7-zip.
>
> It might not be a great idea to store such large archives at GitHub, but
> it is at the moment working and there is not at the present time an
> alternative storage location for these particular builds of the toolchains.

Thanks, I downloaded 7-zip.

I could unzip toolchains

What is the next step?

Thank

ckormanyos, 2022-07-20:
In your case, I recommend to build for the Windows host on the command line as a first step.

The later steps get you building for the embedded AVR controller. But taking an intermediate step to build for a PC will hopefully boost your confidence and provide a much better feeling for how the build system and toolchains all fit together.

So you need to move a directory, unpack a self extracting ZIP and issue a command line.
Let's get started.

You should set up a topology like this (where you simply have the unpacked repositories real-time-cpp (master) and real-time-cpp-toolchains (master). See the image below.

[image]

Now you need to do a few manual steps.

- Move the tools: For this please move the directory `real-time-cpp-toolchains-master/ref_app/tools` to `real-time-cpp-master/ref_app/tools`. See picture below.
- Now we unpack the compiler for x86_64. Navigate to the directory `real-time-cpp-master/ref_app/tools/Util/msys64/usr/local`.
- Now locate the archive called `gcc-11.2.0-x86_64-w64-mingw32.exe`.
- Simply double click on that archive file to unpack the `gcc-11.2.0-x86_64-w64-mingw32` toolchain. Say yes to unpack the toolchain.

Now we will build for Windows host.

1. Open a command shell located in the directory `real-time-cpp-master/ref_app`.
2. Execute the command `.\target\build\build.bat x86_64-w64-mingw32 rebuild`.
3. The build begins.
4. Find the executable file `real-time-cpp-master/ref_app/bin/ref_app.exe` and run it. You should see the simulated PC LED _blinking_.

I found an actual bug in the toolchain. So when building, just click OK on any DLL windows that pop up. I will correct this bug soon.

[image]

[image]

ckormanyos, 2022-07-20:
As a second step, if you like this kind of build, we should get you set up with Visual Studio Community 2022, where you can enjoy the comfort of this build in that GUI.

Cc: @sami1369

jvco98, 2022-07-29:
Hi! I'm having some troubles in setting everything up in a Unix environment. I'm pretty new to it all, but I think I installed GCC correctly using homebrew. I have my Nucleo stm32f446 which I'd like to use but I've been getting some error messages. When following along, I'm stuck here:
[image]
After entering ``` ./target/build/build.sh stm32f446 rebuild
``` i get the error
<img width="564" alt="image" src="https://user-images.githubusercontent.com/22757493/181674595-a5e4d1d9-cb12-4821-9831-3c1788c8b514.png">
Are you familiar with what this would mean? Thanks for any help!

ckormanyos, 2022-07-29:
> Are you familiar with what this would mean? 

Hi Jeff (@jvco98), I'll take a bit more time for this later. Could you please create another issue for this independent point? I would like to also take this opportunity to sue up some more target builds on MacOS in the Ci setup as we progress.

sami1369, 2022-08-09:
> 2\. .\target\build\build.bat x86_64-w64-mingw32 rebuild

Hi again, 
Apologies for the late reply, things were a bit crazy around here these days... 

[image]

My system cant build the command because above error

ckormanyos, 2022-08-09:
> cant build the command because above error

OK. It looks mostly there. The build command is successfully finding GNUmake version 4.2.1. Good job @sami1369.

The next issue is that the make system (within the calls executed in GNUmake) is not finding the x86_64 compiler.

You also need to unpack the compiler, which has been cloned as a self-exctracting zip archive.

Did you remember unpack the mingw x86_64 compiler?

sami1369, 2022-08-10:
Do you mean this?
[image]

> OK. It looks mostly there. The build command is successfully finding GNUmake version 4.2.1. Good job @sami1369.
>
> The next issue is that the make system (within the calls executed in GNUmake) is not [...]

sami1369, 2022-08-10:
Do you mean this? 
[image]

ckormanyos, 2022-08-10:
> Do you mean this?

Yes. My apologies, but the Windows build is actually one of the more complicated ones.

That actually looks good. I'm somewhat surprised the compiler can't be found. Have you tried to run the build again?

ckormanyos, 2022-08-10:
You can also try to manually run the command shown in the failed run above that prints the GCC compiler version.

Something like:

```sh
tools\Util\msys64\usr\local\gcc-11.2.0-x86_64-w64-mingw32\bin\g++ -v
```
```

sami1369, 2022-08-20:
Is there a different way to compile?

ckormanyos, 2022-08-20:
> Is there a different way to compile?

There are numerous ways to compile on Windows, LINUX and MacOS for each supported target.

Each method is described briefly in the documentation at the GitHub page for real-time-cpp, starting around [here](https://github.com/ckormanyos/real-time-cpp#getting-started-with-the-reference-application)

ckormanyos, 2022-08-20:
My first compilation for standalone AVR years ago used ATMEL Studio available now from MICROCHIP. ATMEL Studio is supported in this repository for the reference application.

Some developers use ARDUINO development environment to learn. This is different from the standalone approach used in this repository, but a great way to familiarize yourself with the AVR world.

sami1369, 2022-08-20:
The Arduino development environment is an appropriate choice for me because I tackle with every day. I can use Arduino UNO or ESP32. 
do you have any examples related to the book? It will be very helpful to me. 

I can even work with stm32F1xxx series if you have any suggestions. Even though Arduino is more accessible to me.

Now, It is only important for me to be able to continue your book with exercise.

ckormanyos, 2022-08-23:
Hi @sami1369 I think you are actually rather close to building the reference application either for AVR or for an ARM based core.

Please note that all the book's example projects are actually tested for and developed for an AVR controller, often times using self-made boards in a DIY fashion.

I would like to know how far you got when building the reference application. Where did things stop working?

If you would like to simply install MICROCHIP's ATMEL Studio (available from internet), it can be very straightforward to build the reference application and all the book's examples. Also each target/example is built in the Continuous Integration script.

If you could select one method, I'm pretty sure you are not that far from getting the reference application built and running on a board.

In your opinion, how far did you get and what seems to be failing or breaking down for your build(s)?

sami1369, 2022-08-23:
Hi Christopher,
I want to practice while reading the book and I think that Atmel Studio is
suitable so that I can compile the book's examples and upload them on AVR
boards like Arduino UNO that use an ATmega328.  Could you please let me
know if this is your suggestion?

and can I use book libraries in Atmel Studio?

ckormanyos, 2022-08-23:
> ... please let me know if this is your suggestion?

> and can I use book libraries in Atmel Studio?

Yes @sami1369. This is my suggestion based on our discussion up to this point. This is a very simple and good way to use the book's examples on the Windows platform.

Both the reference application as well as each example project (from the book's chapters) have project workspaces for MICROCHIP's ATMEL Studio.

So if you install ATMEL Studio, you can open any of the `*.atsln` files and build the corresponding project.

For example, if you would like to build the reference application (ref_app), you will find `ref_app.atsln` located [here](https://github.com/ckormanyos/real-time-cpp/tree/master/ref_app).

Each example has its *.atsln` file within its example root folder.

ckormanyos, 2022-08-24:
If you decide to use ATMEL Studio, you should not need any other libraries for these projects other than those that are installed during the standard installation of ATMEL Studio.

The first couple of projects can use the ARDUINO board directly. Later chapters in the book need additional components. Chapter4's examples, for instance, uses additional LEDs that need to be provided.

sami1369, 2022-08-31:
Hi Christopher,

I use this link to run book examples by the link attached.
https://www.instructables.com/How-to-Load-Programs-to-an-Arduino-UNO-From-Atmel-/

Thank you

ckormanyos, 2022-08-31:
> I use this link to run book examples by the link attached.

Thank you for providing this informative link.

Does this mean that you can now use ATMEL Studio to build and successfully flash some of the examples into your ARDUINO device?

sami1369, 2022-08-31:
Yes, it works.

I tested some examples

ckormanyos, 2022-08-31:
>> Does this mean that you can now use ATMEL Studio to build and successfully flash some of the examples into your ARDUINO device?

> Yes, it works.
> I tested some examples

Great! Good work keeping going to success.

I'll close this issue now. If any other issues arise, please feel free to raise another issue any time.

Kind regards, Chris

closed 2022-08-31 by ckormanyos