ckormanyos/real-time-cpp #314: Util::timer timeout timeout method and timer_mask

GitProdEnv, 2022-11-08:
In util::timer there is a timeout method
```
bool timeout() const
    {
      const tick_type delta = my_now() - my_tick;

      return (delta <= timer_mask);
    }
```

If we use a tick_type of std::uint32_t numbers can be represented between 0 and 2^N-1.

I don't fully understand why in the code there is a timer_mask used UINT32_C(2147483647).

If I have a  delta of

```
const std::uint32_t delta = my_now - my_tick;
```

with 
```
const std::uint32_t my_now = 2547483648;
const std::uint32_t my_tick = 100;
```

It should timeout, but the function doesn't. I don't understand entirely why we are using a timer_mask of `(UINTMAX_C(1) << 31) - UINTMAX_C(1)`.

Furthermore `my_now` should be able to be greater or equal to 2^32-1, should it?

ckormanyos, 2022-11-08:
Hi @GitProdEnv thank you for your query.

The timers in `util::timer` are designed to handle overflow of the underlying unsigned tick type. The result of this is that the timer range is basically 1/2 of the numeric range of the underlying unsigned type.

I always have to work out this mathematics from scratch, and I know it works and is easier to grasp for a smaller type such as `uint8_t`. Please give me a chance to work out the math and I'll get back to this thread.

GitProdEnv, 2022-11-09:
Ok after writing the issue, I already thought it would have something to do with under/overflow and that you can only create a timeout value of 1/2 of the numeric range of the unsigned type.

If you use `UINTMAX_C(1) << 31` instead of 

`(UINTMAX_C(1) << 31) - UINTMAX_C(1)` there would be an allowed time difference of delta 2147483648, which is exactly 1/2 of the numeric range of the chosen unsigned type, in this case (2^32)/2. 

```
#include <iostream>

const std::uint32_t timer_mask = 2147483648;
const std::uint32_t my_now = 0;

// Case A
// const std::uint32_t my_tick = 214;
// the unsigned delta type will be greater than the half of the numeric range of the unsigned 32bit type
// and therefore the condition will be 0 which makes sense,
// since the current time 0 tick has not reached the timeout value 214 tick yet.
// it lies within the defined timeout range of `timer_mask`.

// Case B
const std::uint32_t my_tick = 2147483649;
// the difference of my_now and my_tick is greater than half of the numeric range of
// the unsigned type / timer_mask 2147483648 (2^31),
// resulting in a delta  value through the unsigned type: 2147483647 which is
// less than the timer_mask. 
// condition true -> timed out, since the difference of my_now and my_tick
// is larger than the timer_mask (Beware: It is only possible to have a timeout of half
// of the  numeric value of the unsigned type,
// for example my_now: 0 and my_tick: 2147483646, which is less than the 
// time_mask and therefore there is no timeout yet).

int main() {
    const std::uint32_t delta = my_now - my_tick;
    std::cout << delta << std::endl;
    std::cout << (delta <= timer_mask) << std::endl;

    return 0;
}
```
On the other hand if `const std::uint32_t my_tick = 2147483645;`, i.e. is below the timer_tick value, with a `my_now` of 0 we just assume that the timeout has not occured yet, since it lies within 1/2 of the numeric range of the unsigned type.

Therefore wouldn't it be better to just use `UINTMAX_C(1) << 31` for the timer_mask?

ckormanyos, 2022-11-09:
> Ok after writing the issue, I already thought it would have something to do with under/overflow and that you can only create a timeout value of 1/2 of the numeric range of the unsigned type.

Yes. Indeed. I have not yet had time to fully work out the examples. But thank you for diving into this independently.

> Therefore wouldn't it be better to just use `UINTMAX_C(1) << 31` for the timer_mask?

Let's find out...

When doing these kinds of investigations, I have always favored experiments with the 8-bit unsigned type and its range(s). The mathematics with the masking works out exactly the same for 8-bit values. So if we figure it out fully for BYTE, then it works for WORD, DWORD and quad-WORD as well.

Cc: @GitProdEnv

ckormanyos, 2023-08-18:
This is sufficiently described now (I hope), considering

- the discussion in this thread,
- the rGPT refactor in #436
- synchronization with the relevant code snippet in [chapter09_03-001_system_tick.cpp](./code_snippets/chapter09/chapter09_03-001_system_tick.cpp).

ckormanyos, 2023-08-18:
Fixed by #436

closed 2023-08-18 by ckormanyos