ckormanyos/real-time-cpp #451: Build error in chapter02_02 using Windows 11 (x64)

Cesar-Urteaga, 2023-12-02:
I tried to run:

build.bat "Path of bin folder with the gcc-version-avr" avr.

And the results are stored in the "bin"directory.  Nonetheless, I got the error:

"...
libdep.a is either not designed to run on windows or it contains an error.
..."

Can someone tell me how I can run it without errors, please?  

Thank you for your time.

N.B.: It would be useful to have a forum (or a discord/reddit channel) where we can address doubts about this amazing book "Real-time C++).  I know that this book is "gold", but some steps are not detailed enough (at leas for me).  For example, I was unable to build the GNU GCC Cross Compiler following the steps of Appendix C (fourth edition).  And the steps described in the "real-time-cpp-toolchains" repository use different versions from the book and there are some "TBD" notes or patches which I could not follow.  I had to use the "self-extracting archives" of this repository (when I wanted to build the compiler by myself) .  Any help will be appreciated.

ckormanyos, 2023-12-13:
> Can someone tell me how I can run it without errors, please?

Hi @Cesar-Urteaga sorry about the late response on the issue you reported. I have encountered this issue previously and it seems to be related to a (slightly) broken compiler on Windows platform(s).

Could you please tell me which compiler/version you are using?

At the moment, I have forgotten the fix. It is/was simple. I will look through my notes and try to come back here with the workaround.

ckormanyos, 2023-12-13:
> some steps are not detailed enough (at leas for me). For example, I was unable to build the GNU GCC Cross Compiler following the steps of Appendix C (fourth edition). And the steps described in the [...]

These are separate topics and I do appreciate these comments. Cross-building `avr-gcc` on Windows is somewhat challenging and it is a topic that I have not yet fully described/ssimplified. Although I am working on this. I can't exactly promise _when_ simplifications on building the toolchains will become available, but they are on my _todo_ list.

ckormanyos, 2023-12-13:
Regarding the topic

In the particular version of `binutils` used for building the compiler, I suspect, `libdep.a` might be _broken_. See also [here](https://github.com/msys2/MINGW-packages/issues/7890).

I followed the trivial workaround mentioned in that thread [here](https://github.com/msys2/MINGW-packages/issues/7890#issuecomment-774054356), with refinement described in the thread-entry a few posts down [here](https://github.com/msys2/MINGW-packages/issues/7890#issuecomment-774485669).

This workaround is a bit undocumentd, but although i do use `lto`, I do not use `libdep`.

Cc: @Cesar-Urteaga

ckormanyos, 2023-12-13:
See also newly-added notes [here](https://github.com/ckormanyos/real-time-cpp-toolchains/blob/e15a1a3c3501cfdd509babb2da402e9342b39fb5/notes/Building_GCC_11.2.0_for_avr_with_avr-libc.txt#L154).

ckormanyos, 2023-12-13:
See also newly-added issue [here](https://github.com/ckormanyos/real-time-cpp-toolchains/issues/12).

ckormanyos, 2023-12-19:
> the steps described in the "real-time-cpp-toolchains" repository

Indeed, I'm still working out some of the details on the best way(s) to build cross compilers.

Not only this issue, but many influences from previous times motivated the creation of the new repo [avr-gcc-build](https://github.com/ckormanyos/avr-gcc-build). In this repo, exact recipes are given to build `avr-gcc` directly from source. It builds on GCC trunk and v13 at the moment, but is limited to linux `x86_64-linux-gnu` host. I am at the moment not able to successfully cross build `avr-gcc` on linux for host `x86_64-w64-mingw32`. And this is flagged as an [open issue](https://github.com/ckormanyos/avr-gcc-build/issues).

Cc: @Cesar-Urteaga

ckormanyos, 2023-12-25:
Fixed in [ckormanyos/avr-gcc-build](https://github.com/ckormanyos/avr-gcc-build) [issue 13](https://github.com/ckormanyos/avr-gcc-build/issues/13) with change set 7876a1c4c3ccc2e64b101ea45c17f7daad26d616. There we provide a complete recipe for building `avr-gcc` on `x86_64-w64-mingw32`.

closed 2023-12-25 by ckormanyos

ckormanyos, 2023-12-29:
Hi @Cesar-Urteaga here is yet another update. In the (new) repository [ckormanyos/avr-gcc-build](https://github.com/ckormanyos/avr-gcc-build), there is full support for standalone, dependency-free `avr-gcc` toolset on `msys2` (at the moment `avr-gcc`-12.3.0).

The toolchain is provided as an artifact directly from the [CI run](https://github.com/ckormanyos/avr-gcc-build/actions/runs/7357156617).

Full scripts are provided with all details shown and described. The CI is running, but it's going green.

Runs on `ubuntu-latest`focus on more modern GCC versions such as `trunk`.

ckormanyos, 2024-02-11:
Hi @Cesar-Urteaga for the sake of completeness, I have significantly overhauled all compiler builds that I support with a combination of separate repos that build the toolchains of links to moderna areas to get some other toolchains.

These are fully documented in [ckormanyos/real-time-cpp-toolchains](https://github.com/ckormanyos/real-time-cpp-toolchains) and the links therein.