ckormanyos/real-time-cpp #526: C++-ification of RP2040 mcal

ckormanyos, 2024-07-26:
The RP2040 mcal has borrowed significantly from [Chalandi/Blinky_Pico_dual_core_nosdk](https://github.com/Chalandi/Blinky_Pico_dual_core_nosdk). His repo (many thanks for yet more excellent work form @Chalandi) was originally written in no-sdk, standalone C. To mattch styes, the purpose of this issue is to fully C++-ify those original (very helpful) components.

ckormanyos, 2024-07-26:
Cc: @imahjoub this now boots and fully runs in standalone mode for RP2040. The Ozone debug files can be found in the expected target-specific area.

ckormanyos, 2024-07-26:
Change the `Mcal` files one-by-one to be integrated in the normal C++ repo-style.

ckormanyos, 2024-07-26:
And when it is done, remember to remove the temporary CMSIS license docs notes.

imahjoub, 2024-07-26:
> Cc: @imahjoub this now boots and fully runs in standalone mode for RP2040. The Ozone debug files can be found in the expected target-specific area.

thank you @ckormanyos

commit ab7e53b6 (2024-07-28): Merge pull request #527 from ckormanyos/cpp_ification_of_rp2040

Fixes #526 via C++-ification of rp2040
closed 2024-07-28 by ckormanyos